MickTheMechanic/Shiftlight #2: Tach Input

adamgav5000, 2020-12-03:
Hi Michael,

I'm really looking forward to building and running this project. I have a few questions about the incoming tach signal...

Do you know where your vehicle reads its tach signal from? Is it the negative side of the coil?
You say the resistors etc are based off of a 12v signal. Is this signal a clean 12v square wave?

I am trying to get this to work with a YX160 pitbike engine which does not have an original tachometer signal. I am planning on taking a reading from the negative of the primary side of the coil. I believe this is susceptible to voltage surges of 200V. I am yet to check out the signal with an oscilloscope, but have found other peoples readings that suggest it may be a sawtooth pulse rather than a nice square pulse. 

Can you suggest a method to reduce and clean the voltage to a level that can be applied to the base of the transistor.
What is the target voltage and current that is needed to be sent to the base?

My coding ability is non existent as is my circuit design knowledge, so any help is much appreciated!

Kind Regards
Adam

michaelwilkinson1992, 2020-12-03:
Hi, this particular circuit is designed to use with a 12v sqare wave signal. 

The signal is the tacho output from the ECU.

You could definatly use the coil negative as a signal, but you would need to update the code a little bit to match the frequency of the coil signal. This can be easily calculated, all you need to know is how often the coil fires for one engine rotation.

But... I guess your bike does not have electronic ignition, rather a magneto style ignition coil. In this case you will need to do some research on how to convert this to some kind of signal that is not greater than 5v ( It shouldnt matter if you have a sawtooth, square wave, or sinus wave signal, as the arduino will still register the top and bottom of the signal as high/low).

The important thing is the lows are 0v and the highs are somewhere in the ballpark of 3-5v.

I reccomend reading the signal first with an occilliscope, if your signal does not exceed 15v the existing circuit should work.
If you are getting high induction spikes then you will need to clean up the signal somehow.

adamgav5000, 2020-12-13:
Sorry for the slow reply. Thank you for the information!

closed 2021-11-21 by MickTheMechanic