MickTheMechanic/Shiftlight #3: Trying to run on a 6 cylinder

cpatersontn, 2020-12-07:
Mick,
Thanks for posting this project. We're looking forward to getting this running on our race car. We have a 1991 BMW 325i running the 6 cylinder engine. We've built the board as you specified and uploaded a modified version of your code. We are pulling the tach signal from the ECU wire that feeds the factory in-dash tach. The factory tach is working fine so we know it's getting good signal. The engine is firing 3X per revolution so we are assuming the frequency (at a given rpm) is (RPM/60)*3. We are testing at lower RPMs (don't want to upset the neighbors with rev'ing it to redline :)) so we calculated the lower RPMs between 1500 and 3100 (75Hz-155Hz). It idles at 800 RPM. 

Once we start the car, the start-up sequence works and they flash one by one and then go off. However, regardless of how high we rev it, nothing else lights up. It's like it does the start-up sequence then somewhere it gets hung-up on lighting the LEDs (0,0,0). Stumped. 

Here's the modified code:

//Shift Light V1.5

#include <Adafruit_NeoPixel.h>

#define PIN            6  //LED Data Pin
#define NUMPIXELS      16  //number of leds connected
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

//Calculate the frequency of the tacho signal you are using and adjust the following values accordingly
//example: for a standard tacho signal on a 4 cylinder engine the frequeny is equal to the RPM divided by 30
const unsigned int onFrequency = 20;  //startup sequence, value represents engine speed higher than cranking and lower than idle 
const unsigned int minFrequency = 75; // minimum frequency to begin turning on LEDs 
const unsigned int maxFrequency = 155; // maximum frequency in normal range, after this value shift flash will occur
const unsigned int shiftFrequency = 165; // frequency range from max to shift when shifting should happen, after this value overrev flash will occur

//Set the color for each LED in the neo pixel strip
//Colors are set using an RGB value ranging from (0,0,0) to (255,255,255)
const uint32_t tachColor[NUMPIXELS] = {
  Adafruit_NeoPixel::Color(0, 120, 0),//green
  Adafruit_NeoPixel::Color(0, 120, 0),
  Adafruit_NeoPixel::Color(0, 120, 0),
  Adafruit_NeoPixel::Color(15, 105, 0),
  Adafruit_NeoPixel::Color(30, 90, 0),
  Adafruit_NeoPixel::Color(45, 75, 0),
  Adafruit_NeoPixel::Color(60, 60, 0),//orange
  Adafruit_NeoPixel::Color(75, 45, 0),
  Adafruit_NeoPixel::Color(90, 30, 0),
  Adafruit_NeoPixel::Color(105, 15, 0),
  Adafruit_NeoPixel::Color(120, 0, 0),//red
  Adafruit_NeoPixel::Color(0, 0, 120),//blue
  Adafruit_NeoPixel::Color(0, 0, 120),
  Adafruit_NeoPixel::Color(0, 0, 120),
  Adafruit_NeoPixel::Color(0, 0, 120),
  Adafruit_NeoPixel::Color(0, 0, 120),
};

//Set the frequency when each LED should turn on
//First LED turns on at minFrequency
const unsigned int lightShiftFreq[NUMPIXELS] = {
  minFrequency,
  80,
  85,
  90,
  95,
  100,
  105,
  110,
  115,
  120,
  125,
  130,
  135,
  140,
  145,
  150,
};

bool hasStartupSequenceRun = false; // only run startup sequence one time
const byte tachPin = 2;
unsigned long igfreq;

void setup() {
  pixels.begin(); // This initializes the NeoPixel library.
}

void loop() {

  float ighigh, iglow;
  unsigned long igcal1, igcal2;

  //measure period of tach signal
  ighigh = pulseIn(tachPin, HIGH);
  iglow = pulseIn(tachPin, LOW);

  igcal1 = 1000 / ((ighigh / 1000) + (iglow / 1000));

  //do it again
  ighigh = pulseIn(tachPin, HIGH);
  iglow = pulseIn(tachPin, LOW);

  igcal2 = 1000 / ((ighigh / 1000) + (iglow / 1000));

  //to filter out some noise, we only consider our measurement valid if they are similar in value, we accept the average.
  if ((igcal1 - igcal2) < 8) {
    igfreq = (igcal1 + igcal2) / 2;
  }

  if (hasStartupSequenceRun == false) {
    if (igfreq > onFrequency) {
    //run start sequence
    //LEDs will light up, flash and light out upon starting the engine.
      for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        delay(50);
      }
      for(int a=0; a<10; a++) {
        pixels.fill(pixels.Color(0, 0, 120));
    	pixels.show();
    	delay(20);
    	pixels.fill(pixels.Color(0, 0, 0));
    	pixels.show();
    	delay(20);
       } 
       for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        }
       for ( int i = NUMPIXELS-1; i >= 0; --i) {
        pixels.setPixelColor(i, pixels.Color(0, 0, 0));
        pixels.show();
        delay(50);
      }
      hasStartupSequenceRun = true;
      pixels.fill(pixels.Color(0, 0, 0));
      pixels.show();
    }
      if (igfreq < onFrequency) {
    //resets hasStartupSequenceRun to false if engine stops but Arduino remains powered, 
    //startupsequence will rerun upon restarting the engine
	hasStartupSequenceRun = false;
    }
  }
	
  if (igfreq < maxFrequency) {

    // normal operating range
    for ( int i = 0; i < NUMPIXELS; ++i) {
      if (igfreq > lightShiftFreq[i]) {
        pixels.setPixelColor(i, tachColor[i]);
      }
      else {
        pixels.setPixelColor(i, pixels.Color(0, 0, 0));
      }
    }
    pixels.show();
  }
  else if (igfreq >= maxFrequency && igfreq < shiftFrequency) {
    //shift flash
    //default color=blue
    //to change shift flash color, edit RGB value in following line
    pixels.fill(pixels.Color(0, 0, 120));
    pixels.show();
    delay(20);
    pixels.fill(pixels.Color(0, 0, 0));
    pixels.show();
    delay(20);
  }

  else if (igfreq >= shiftFrequency) {
    //overrev flash
    //default color=red
    //to change overrev flash color, edit RGB value in following line
    pixels.fill(pixels.Color(120, 0, 0));
    pixels.show();
    delay(20);
    pixels.fill(pixels.Color(0, 0, 0));
    pixels.show();
    delay(20);
  }
}

MickTheMechanic, 2020-12-07:
If you have an ocsilloscope please check your Signal wire and confirm you have a 12v square wave (I am not familiar with BMW Tacho Signal, it’s possible it’s a 5v Signal).

Does your car have a single coil with a distributor? If so, please try using coil negative.

Report back with results.

cpatersontn, 2020-12-07:
We don't have an ocsilloscope. If it's a 5V signal, then what do we need to modify (hardware and software)? 

Yes, single coil with distributor. We tried the coil negative and it was very noisy.... LEDs jumped all over the place.

MickTheMechanic, 2020-12-07:
If its a 5v signal then you can feed it directly into the arduino, bypassing the transistor. But if its not 5v and you do that, you'll fry your arduino.

Do you maybe have a video showing what it looks like when using coil negative?

MickTheMechanic, 2020-12-07:
You could even check with a multimeter, if its a 12v signal you should read about 6v on the signal wire with engine running. If its a 5v signal you should read about 2-3v

cpatersontn, 2020-12-08:
Tach signal from ECU is 7.9V (set to DC on multimeter). 

Here's the shift light attached to the negative on the coil. It always goes through the start-up sequence then just flashes red as at idle if it's over-reving. Our shiftFrequency is set to 165Hz just for testing but that should equate to 3300rpm if our equation is correct for this 6-cylinder (RPM/60)*3)

Link to video: [https://www.facebook.com/chris.paterson.399/videos/10219498587075256/](url)

MickTheMechanic, 2020-12-08:
Link doesnt work.

Are you able to test on another car? 
Perhaps there is a problem with the circuit.

Have you tried using the original code, unchanged?

MickTheMechanic, 2020-12-08:
Can you upload photos of your circuit?

cpatersontn, 2020-12-08:
Try copying and pasting the linked address instead of clicking on it

Ok on tach signal

Can't try it on any other vehicle at this time

Here's pics of the PCB and LED. My first ones I've ever built so definitely can use any pointers or corrections if something doesn't look right

[image]
[image]
[image]
[image]
[image]
[image]

MickTheMechanic, 2020-12-09:
I just had a look at the video, the red flashing appears to be the overrev flash. 
Do you only have this problem when using coil negative, or also with the Tacho Signal wire from the dash?

cpatersontn, 2020-12-09:
When we run the unmodified program, we have red blinking lights on the negative coil and tach signal

MickTheMechanic, 2020-12-09:
I just updated my previous comment

cpatersontn, 2020-12-09:
Thanks. We'll try a few experiments and see if changing values higher gives a different response. 

In this section of the program, it looks like it is measuring the tach period and then doing some function to scale that number. Why is it dividing everything by 1,000? 

//measure period of tach signal
ighigh = pulseIn(tachPin, HIGH);
iglow = pulseIn(tachPin, LOW);

igcal1 = 1000 / ((ighigh / 1000) + (iglow / 1000));

MickTheMechanic, 2020-12-09:
Because the pulseIn() doesnt read Hz, rather it measures how many microseconds between pulses. From this we have to calculate Hz.
This code is run twice, both values are compared, and if similar the average of both values is used (helps cut down unwanted noise).

MickTheMechanic, 2020-12-09:
Also had a closer look at your board, the circuitry look ok, seems everything is connected in the right spot. Also confirmed by the fact that the start up sequence runs.

cpatersontn, 2020-12-09:
Ok, thanks on explaining the pulseIn() code. Would you expect a magnitude difference between different makes/models/# cylinders on the actual time between pulses which would affect the calculated Hz? For instance, your original code has 123Hz @ 3700 RPMs.... would it be unexpected if the Hz calculation for our BMW 6-cylinder is a magnitude different? e.g. 2000 Hz (or higher) @ 3700 RPMs ?

MickTheMechanic, 2020-12-09:
The time between pulses and the frquency are directly related to one another. the time between one pulse and the next is called a period and the frequency (hz) is periods per second.

I believe the code to be in order, as other members from the Livetodai.com forum have used this design/code with success.

I expect the frequency of your tacho signal should be as you described, RPM/60=revolutions per second, multiplied by how many times the coil fires per revolution (6 cylinder should fire 3 times per rotation).

Unfortunately Im not quite sure how your signal looks (it would take 5 min with an occilloscope to check)

If you do not have these resources available (e.g ocsilloscope) then you will have to do some trial and error to find out which frequencies correspond to which RPM's.

I can suggest using your device to measure the frequency

Simply add the following line to void loop()

Serial.println (igfreq);

then open the arduino IDE serial plotter (Tools -> Serial plotter), while connected to the arduino with the engine running, and your arduino will basically function like a make shift oscilloscope, giving you a live readout of the tacho signal frequency.

Hope this helps.

cpatersontn, 2020-12-12:
Mick,
Still no luck with the board. I tried a few things:
1. Updated your original code by adding **Serial.println (igfreq);** directly after the **void loop()** line. Opened the Serial Plotter and the screen was just blank. Tried to mess with the baud rate and menu options on the lower right but the screen never showed anything. 

2. I connected the tach signal from the negative coil and then tried the ECU tach signal line in to see if it made a differernce. Changed the ShiftFrequency to 1000+ which then stopped it blinking red but then the LEDs wouldn't light or would light erratically regardless of what I changed the values to. Just no pattern that I could discern.

3. As long as OnFrequency was below 30 then the start-up sequence always started.

MickTheMechanic, 2020-12-12:
Sorry, I did not explain myself properly last time.
download a fresh copy of shiftlight.ino

Add the following line to void startup:
```

Serial.begin(9600);
 
```

Add the following line to void loop, directly after  igfreq = (igcal1 + igcal2) / 2; }
```
Serial.println (igfreq);

```

Open serial plotter, make sure baud rate is set to 9600

Start the engine and take note of the frequency.

report back with results.

cpatersontn, 2020-12-13:
Here's the modified code that we tried today and the screen shot of the Serial Plotter. We have this connected to the tach-in signal from the ECU that is also connected to the factory (which is working correctly). Serial Plotter now shows up but no signal shown. The blinking start-up signal works everytime we start the car but no other LEDs after that regardless of RPM. We disconnected the board from the ECU wire just to see what would happen and there was no start-up sequence as expected. Just verifying that it wouldn't go through that sequence without a signal. Weird that it gets that signal to start-up but then nothing after that.

Modified Code: 

//Shift Light V1.5

#include <Adafruit_NeoPixel.h>

#define PIN            6  //LED Data Pin
#define NUMPIXELS      16  //number of leds connected
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

//Calculate the frequency of the tacho signal you are using and adjust the following values accordingly
//example: for a standard tacho signal on a 4 cylinder engine the frequeny is equal to the RPM divided by 30
const unsigned int onFrequency = 20;  //startup sequence, value represents engine speed higher than cranking and lower than idle
const unsigned int minFrequency = 123; // minimum frequency to begin turning on LEDs
const unsigned int maxFrequency = 235; // maximum frequency in normal range, after this value shift flash will occur
const unsigned int shiftFrequency = 250; // frequency range from max to shift when shifting should happen, after this value overrev flash will occur

//Set the color for each LED in the neo pixel strip
//Colors are set using an RGB value ranging from (0,0,0) to (255,255,255)
const uint32_t tachColor[NUMPIXELS] = {
  Adafruit_NeoPixel::Color(0, 120, 0),//green
  Adafruit_NeoPixel::Color(0, 120, 0),
  Adafruit_NeoPixel::Color(0, 120, 0),
  Adafruit_NeoPixel::Color(15, 105, 0),
  Adafruit_NeoPixel::Color(30, 90, 0),
  Adafruit_NeoPixel::Color(45, 75, 0),
  Adafruit_NeoPixel::Color(60, 60, 0),//orange
  Adafruit_NeoPixel::Color(75, 45, 0),
  Adafruit_NeoPixel::Color(90, 30, 0),
  Adafruit_NeoPixel::Color(105, 15, 0),
  Adafruit_NeoPixel::Color(120, 0, 0),//red
  Adafruit_NeoPixel::Color(0, 0, 120),//blue
  Adafruit_NeoPixel::Color(0, 0, 120),
  Adafruit_NeoPixel::Color(0, 0, 120),
  Adafruit_NeoPixel::Color(0, 0, 120),
  Adafruit_NeoPixel::Color(0, 0, 120),
};

//Set the frequency when each LED should turn on
//First LED turns on at minFrequency
const unsigned int lightShiftFreq[NUMPIXELS] = {
  minFrequency,
  130,
  137,
  144,
  151,
  158,
  165,
  172,
  179,
  186,
  193,
  200,
  207,
  214,
  221,
  228,
};

bool hasStartupSequenceRun = false; // only run startup sequence one time
const byte tachPin = 2;
unsigned long igfreq;

void setup() {
  pixels.begin(); // This initializes the NeoPixel library.
}

void loop() {

  Serial.begin(9600);

  float ighigh, iglow;
  unsigned long igcal1, igcal2;

  //measure period of tach signal
  ighigh = pulseIn(tachPin, HIGH);
  iglow = pulseIn(tachPin, LOW);

  igcal1 = 1000 / ((ighigh / 1000) + (iglow / 1000));

  //do it again
  ighigh = pulseIn(tachPin, HIGH);
  iglow = pulseIn(tachPin, LOW);

  igcal2 = 1000 / ((ighigh / 1000) + (iglow / 1000));

  //to filter out some noise, we only consider our measurement valid if they are similar in value, we accept the average.
  if ((igcal1 - igcal2) < 8) {
    igfreq = (igcal1 + igcal2) / 2;
    Serial.println (igfreq);
  }

  if (hasStartupSequenceRun == false) {
    if (igfreq > onFrequency) {
    //run start sequence
    //LEDs will light up, flash and light out upon starting the engine.
      for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        delay(50);
      }
      for(int a=0; a<10; a++) {
        pixels.fill(pixels.Color(0, 0, 120));
    	pixels.show();
    	delay(20);
    	pixels.fill(pixels.Color(0, 0, 0));
    	pixels.show();
    	delay(20);
       } 
       for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        }
       for ( int i = NUMPIXELS-1; i >= 0; --i) {
        pixels.setPixelColor(i, pixels.Color(0, 0, 0));
        pixels.show();
        delay(50);
      }
      hasStartupSequenceRun = true;
      pixels.fill(pixels.Color(0, 0, 0));
      pixels.show();
    }
      if (igfreq < onFrequency) {
    //resets hasStartupSequenceRun to false if engine stops but Arduino remains powered, 
    //startupsequence will rerun upon restarting the engine
	hasStartupSequenceRun = false;
    }
  }
	
  if (igfreq < maxFrequency) {

    // normal operating range
    for ( int i = 0; i < NUMPIXELS; ++i) {
      if (igfreq > lightShiftFreq[i]) {
        pixels.setPixelColor(i, tachColor[i]);
      }
      else {
        pixels.setPixelColor(i, pixels.Color(0, 0, 0));
      }
    }
    pixels.show();
  }
  else if (igfreq >= maxFrequency && igfreq < shiftFrequency) {
    //shift flash
    //default color=blue
    //to change shift flash color, edit RGB value in following line
    pixels.fill(pixels.Color(0, 0, 120));
    pixels.show();
    delay(20);
    pixels.fill(pixels.Color(0, 0, 0));
    pixels.show();
    delay(20);
  }

  else if (igfreq >= shiftFrequency) {
    //overrev flash
    //default color=red
    //to change overrev flash color, edit RGB value in following line
    pixels.fill(pixels.Color(120, 0, 0));
    pixels.show();
    delay(20);
    pixels.fill(pixels.Color(0, 0, 0));
    pixels.show();
    delay(20);
  }
}
[image]

MickTheMechanic, 2020-12-13:
Here is what you have;
```

igfreq = (igcal1 + igcal2) / 2;
Serial.println (igfreq);
}

```

Try this:
```

igfreq = (igcal1 + igcal2) / 2;
}
Serial.println (igfreq);

```

MickTheMechanic, 2020-12-13:
Please upload your code, you can drag and drop the .ino file into the comment box.

MickTheMechanic, 2020-12-13:
You also have Serial.begin inside void loop, it must be inside void setup!

cpatersontn, 2020-12-13:
So for Serial.begin, should it look like this in this seciton of the program?

bool hasStartupSequenceRun = false; // only run startup sequence one time
const byte tachPin = 2;
unsigned long igfreq;

void setup() {
Serial.begin(9600);
pixels.begin(); // This initializes the NeoPixel library.
}

void loop() {

float ighigh, iglow;
unsigned long igcal1, igcal2;

MickTheMechanic, 2020-12-13:
Yes exactly!

Here, stripped down the code to a very basic tester for checking the frequency.

Upload this to your Arduino, open serial plotter, start engine. 

Report back with results.

(And please upload the your code directly. Send to a .zip folder and drag and drop to the comment box. I want to check it, some characters are displayed differently when you just copy paste the text!)

[Test_Frequency.zip](https://github.com/MickTheMechanic/Shiftlight/files/5685520/Test_Frequency.zip)

cpatersontn, 2020-12-13:
[test_light1.zip](https://github.com/MickTheMechanic/Shiftlight/files/5685567/test_light1.zip)

MickTheMechanic, 2020-12-13:
https://www.tinkercad.com/things/7d5KEDsRvIV-shiftlight-code/editel?sharecode=V9_L_blID8e-IG5Z-QfdU7xAY3IYFB3_7xq6teOL588

Here is the design on tinkercad.
A signal generator provides the tacho signal.
You can see what happens to the LEDs when you change the frequency on the signal generator, and you can take a look at the serial monitor and see the frequency being displayed.

Play around with it.

Here is a copy of the code I used in Tinkercad. Its the one you uploaded, I only changed the 2 things I mentioned above (Serial.begin in void startup, and Serial.print outside of the bracket.)

Code works, 100% (as confirmed by simulation in tinkercad).
You just gotta find the right frequencies (or maybe your tacho signal is not compatible, but I doubt it).

[test_light1.1.zip](https://github.com/MickTheMechanic/Shiftlight/files/5685617/test_light1.1.zip)

cpatersontn, 2020-12-13:
Thanks, will give this new code a try this week

cpatersontn, 2020-12-19:
New experiment this weekend:
1. First connected signal wire to coil (negative) and ran the testlight1 program. Received data which was great. Interesting that the signal bounces high/low. Pics and videos attached. 
[image]

https://user-images.githubusercontent.com/75594078/102696412-acfc3400-41f3-11eb-9240-888e8be12097.MOV

https://user-images.githubusercontent.com/75594078/102696413-ad94ca80-41f3-11eb-82f6-466197f2913a.MOV

cpatersontn, 2020-12-19:
Next we ran the signal wire to the ECU. We confirmed that the signal is good as this is the same wire to the factory tach which was reading correctly. Intersting that we got zero data. Just a thin line at 0.00. Why would we get no data from the ECU signal? We've confirmed that the voltage is ~7.9VDC. Any ideas based on these experiments what could be going on with the coil and ECU signals? 
[image]

MickTheMechanic, 2020-12-19:
Did you have the Arduino grounded to the car? You must have one of the gnd Pins on the Arduino connected to the car chassis. Normally this is the case if you are powering the Arduino from 12v via the LM7805, but i am guessing you just had it plugged into the laptop? That’s the only thing I can think of at the moment

cpatersontn, 2020-12-19:
Ground- yes, I have both GND pins on both sides of the Arduino chip wired to ground on the board which is wired to the chassis. 

Wiring is positive (red) 12VDC from ignition. Goes through the LM7805 and confirmed 5VDC output from the LM7805. Ground (black...same side as the red positive wire) to the car chassis. Confirmed ground continuity to both GND pins and to board to chassis. Signal wire either from the ECO (tach) wire or directly from the coil negative. 

Why would the coil give a signal through the Serial Plotter function but nothing when connected to the ECU? 

Also, it looks like the signal from the coil is pretty erratic. Big highs and lows. Does that look normal and is it usable from the coil?

MickTheMechanic, 2020-12-19:
Its reading about 40kHz, but the signal should only be 100-200 Hz. There is some kind of interferance there somewhere. 

-Double check your circuit, there is possibly something amiss there thats allowing interferance to be read as pulses.

-Check the routing of the signal wire, it should not be run along side any kind of signal wires for valves that are powered by PWM (things like idle control valve, AGR and so on)

-If you have the posibilitiy, perform a test without the alternator connected, its possible the alternator voltage is producing a ripple that is being picked up by the arduino.

It seems very strange that the arduino is reading such a high frequency, and not reading anything on the tacho signal. Perhaps the arduino has been damaged in some way, possibly by a voltage surge from the coil (induction voltage when the points/coil driver opens)

I have attached some extra photos so you can check your circuit.
It is also important that you are using the same kind of transistor (PNP, not NPN)

[image]
[image]

MickTheMechanic, 2020-12-19:
Please provide accurate information about your car, make model, build date, engine, ECU. I will take a look at work on monday and see if I can find a wiring diagram/somewhere where you can pick up a signal

cpatersontn, 2020-12-19:
Thanks Mick. Will check those this weekend. 

Car is a 1991 BMW 325i, 6 cylinder. Stock ECU. 

Here's my part list:
330k ohm 5W resistor
10k ohm 5W resistor
47k ohm 3W resistor
PNP Transistor ON Semi Bipolar Transistors
Heat sink
5 volt regulator 
1000uF 20% 16V Vishay Aluminum Electrolytic Capacitor 
Arduino Nano
Adafruit LEDs

MickTheMechanic, 2020-12-21:
If you have the 20 pin style BMW diagnostic port, try Pin 1 (listed as Drehzahl in below picture)

[image]

MickTheMechanic, 2020-12-21:
Does your car have one ignition coil, or 6. 
please provide a picture of how you have connected to the ignition coil.

cpatersontn, 2020-12-22:
Our car has one standard cylindrical coil. Here is the attachment point to the negative side. I ran it directly to the Arduino without any other wires around it. 
[image]

MickTheMechanic, 2021-02-24:
Did you get it working?

cpatersontn, 2021-03-01:
Unfortunately, no. The issue was either I got zero signal from the factory tach wire from the ECU or the signal from the coil was too noisy and just bounced around the signal. Thanks for your help and understanding the code and circuit. May just need to build another one from scratch to see if it was a hardware issue.

MickTheMechanic, 2021-04-06:
Hi, 

I was working on a project recently which reminded me of your problem.

The project used some transistors, and I came to learn the hard way that not all transistors have the same pinout. E,C, and B pins vary from manufacturer to manufacturer. I even had two 2N2222A transistors with different pin layout, the only difference was a prefix printed on the housing.

I thought maybe this could be the problem. If you are still interested in getting this project up and running it might be a good place to start with the troubleshooting. You would need to download the datasheet for the particular transistor you have (should be available from the seller website) and confirm the pin layout matches how you have it connected.

cpatersontn, 2021-04-06:
Interesting..... will need to check

Would a higher resistor value cause issues with this circuit?

MickTheMechanic, 2021-04-06:
Yes it could have an influence. Transistors are not actually switched by the amount of voltage, but rather by the amount of current that runs through the base. This current is limited by the resistor connected to the base, so a different value resistor could have a negative effect.

also Different transistors have Different characteristics regarding switching speed, switching current, and so on.

I would first of all check the datasheet and make sure you have the pins connected correctly (in my last project I was pulling my hair out trying to find the problem, turns out the transistor I used that had „similar properties“ also had a different pin layout!).
If that doesn’t work I would swap it out with one that has the same part number as listed in the parts list, as the resistors have been sized to suit this one.

just some things for you to consider if your interest in the project is reignited sometime.

cpatersontn, 2021-04-06:
Ok,I’ll need to double check the resistors too as I think one or two are close tot the right values but not exactly to the spec of your original design. I’ve got a few of the transistors too that I swap out. We really want to get this working this summer so we’ll be trying a few things to see

MickTheMechanic, 2021-04-07:
The big 3watt resistors are not necessary, you can also use normal 0.6w or 0.25w resistors without any problems.

MickTheMechanic, 2021-06-05:
I found the solution to your problem, it’s documented in #4 .

if you have time you should check it out

cpatersontn, 2021-06-07:
Thanks Mick. Will take a look

cpatersontn, 2022-01-09:
Hi Mick,

Not sure how I missed this email but glad I now saw it. I’m going to check my original circuit to see where it needs to be modified to this new one your drew. Do you have a complete circuit draw with your new modification like the original one you posted?

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

cpatersontn, 2022-01-29:
Mick,
We’re bench testing the modified board. Should the LED start up sequence happen every time we plug in the USB cable to the board and computer?

MickTheMechanic, 2022-01-29:
yes the startup sequence should run under the following conditions:
1. the device is switched on/the engine is started
2. the device is already on, if the engine stalls, and is restarted

cpatersontn, 2022-01-30:
Ok…. Working on a few tweaks. Interesting LED behavior. LEDs on on one pixel board stay lit but the other board goes through the start-up sequence… but not every time when connected to the laptop. Any idea what would cause this strange LED behavior?

MickTheMechanic, 2022-01-30:
Hi, I think I have made a mistake in my previous explanation (Its been a long time since I designed this).

The startup sequence should only run once the arduino detects an engine start. The arduino can calculate the input frequency with just 2 measurements, so its possible sometimes when plugging the arduino into to cable the arduino measures a very short spike and thinks this is the input frequency. This happens because the tacho pin is floating, and not in a high or low state. Once its installed in your car this pin will always be in either a high or a low state so this should no longer happen.

The more interesting question is what is going on with your LEDs, this is the real problem at hand.

Some things for you to test (in no particular order, justv some stuff that comes to mind):
1. Upload a test sketch to the arduino in order to test the function of the LEDs. There should be a test sketch included in your Arduino IDE if you fo to file -> examples -> adafruit_neopixel -> strand test. Then adjust the definitions to match your data pin, number of LEDs and so on. This will eliminate a hardware problem.

2. if that doesnt work Check the data sheet of your LED strips. Make sure the specifications of your particular LEDs match the definitions in the first few lines of code (The type (eg. NEO GRB) and the frequency (e.g KHZ_800 = 800 Hz) : 
 ```

#define PIN            6  //LED Data Pin
#define NUMPIXELS      16  //number of leds connected
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

```
I am pretty sure these are standard specs, but its worth a look.

2. Check your connections, make sure both GND pins are connected to ground, make sure your PWR pin is connected to the 5v supply, and make sure the data line is connected to pin 6 via the 330 ohm resistor. The resistor on the data line is pretty important if I remember correctly.

3. test the device by powering it from the LM7805, maybe this is causing some kind of issue when it is being fed power from via USB without any input voltage.

DeanDerBaum, 2022-03-18:
Hello there,
have you already solved the problem and made it run @cpatersontn ?
Thanks!

M52byfix, 2022-09-24:
@MickTheMechanic Hello mate. I have a 6-cylinder BMW car. This topic is different from the car that is being processed. It uses 6 coils. 20-pin diagnostics I have the connection. I want to do this project. How can you help me

cpatersontn, 2022-10-03:
> Hello there, have you already solved the problem and made it run @cpatersontn ? Thanks!

Sorry for the very tardy reply! No, we still can't get it to work. We'll be messing with it again at the end of the season and may just rebuild the entire board and start over.

cpatersontn, 2022-10-11:
We rewired to the new schematic and need to change the program variables to match our 6-cyl rpms.