MickTheMechanic/Shiftlight #4: Trying to get the project running

Lakshitha747, 2021-05-28:
Mike,

Thanks for the project. Trying to get the code working for my honda (4 cylinder). Did everything according to your diagram without using the LM7805 as im powering my Arduino Uno from my laptop for now. This works but I'm getting frequency values when my ECU wire is not connected and the the idle frequency is showing in the high 40000. If you could find the time to help me that would be really great. Im sort of new to this and I have no idea whats going on

Thank you

MickTheMechanic, 2021-05-28:
Very strange indeed... the same problem was reported in #3 

I will need to look into this further. I know of several other people who have successfully built the project without issues, so I am fairly confident it’s not an issue with the code. 

Can you provide more information, like what you are using as a signal, does the bug also appear when you have the signal wire connected etc etc...

Lakshitha747, 2021-05-28:
Thanks for your help,

Yeah its the same when its connected. It cant be an issue with the code cause the first time I tested it I got the proper values. The idle frequency was at 30Hz and when pin was removed it was showing 0Hz

Lakshitha747, 2021-05-28:
Working on getting you the pictures. Actually didn't expect a reply so quickly

MickTheMechanic, 2021-05-28:
I found the problem, my wiring diagram is wrong... I will post an updated version in a few moments

Lakshitha747, 2021-05-28:
Thank you for the help.

MickTheMechanic, 2021-05-28:
[image]

please try this configuration with the PNP transistor and let me know how it works. 

Also make sure your Arduino gnd pin has a connection to the car body when you are powering via the laptop.

Lakshitha747, 2021-05-28:
Alright thanks. I'll let you know how this goes

MickTheMechanic, 2021-05-30:
Did it work out for you?

Lakshitha747, 2021-05-30:
Sorry waiting on the transistors. Country is on lock down bit hard to get the stuff. I'll update the forum as soon as I get the parts and build the circuit. Is there another alternative to the transistor that you know of other than the PNP transistor #2N3906?

MickTheMechanic, 2021-05-30:
Do you have any other PNP transistors?

Lakshitha747, 2021-05-30:
s8550, s8050 transistor

MickTheMechanic, 2021-05-30:
You can try the S8550, it would probably work.

Lakshitha747, 2021-05-30:
Alright. I'll try with the S8550 and let you know

MickTheMechanic, 2021-06-03:
Any luck?

Lakshitha747, 2021-06-03:
Trying it right now with a S8050 transistor (cant seem to find my S8550 transistors). Trying to print igfreq but the serial monitor shows nothing. I have a random println in the loop and even that doesn't print. Can this be expected when running with this transistor?

Lakshitha747, 2021-06-03:
Ok so now its printing a bunch of zeros
 
[image]

Lakshitha747, 2021-06-03:
Yeah now I took the ground off and now its showing 50s 49s. When the revs increase it increases by like +1 not much and on idle and even when the vehicle is off its saying a constant 50. Looks like the transistor. I'll have get new ones

[image]

Lakshitha747, 2021-06-05:
Well my transistors finally came and I tried it with the 2N3906 and it works. Thanks for all your help with this. Just one more thing, when I do use this with the 12v, I can use a step down converter to reduce the voltage and power the Arduino and the rest of the circuitry will stay the same?

[image]

The frequencies are set pretty low

MickTheMechanic, 2021-06-05:
Yes, you can use either a stepdown or a linear regular, and the rest will stay the same. 

Which wiring diagram worked for you in the end?

Lakshitha747, 2021-06-05:
This one

 [image]

Lakshitha747, 2021-06-05:
I'll close this issue now. Again thanks for your help with this.

closed 2021-06-05 by Lakshitha747

JeFeRoJo, 2021-11-11:
> Well my transistors finally came and I tried it with the 2N3906 and it works. Thanks for all your help with this. Just one more thing, when I do use this with the 12v, I can use a step down [...]

Hello. 

I just created an account here to ask you how you have implemented the display with the RPM counter.
I am also trying to get the project to work on my Honda (EUDM Civic EK3) and am going to test it again with the new published diagram.

Thanks!