MickTheMechanic/Shiftlight #5: Trying to make it run (for a long time)

JeFeRoJo, 2021-11-19:
Hello!

First of all, thank you for sharing your project. I was looking to mount something similar in a simple way and this is great.
But not everything is happiness. I've been stuck with your project for almost a year. I can't get it to work.

A few days ago I took up the project again and saw that it updated the circuit (the position of the PNP transistor).

I have reassembled it following the new scheme. Same components.
The code has not been modified and I think that, could the problem be there?

My car is a Honda Civic EK 1996. D15Z6 engine from the European market. 
I extract the signal from the cable that connects the original tacho/speedometer (blue wire in my case).
I'm just getting uncontrolled blinks. I attach a small video to show it:

https://www.youtube.com/watch?v=rK0u13ykE58

Could it be wrong values ​​in the code for the engine?
Any help would be appreciated. I would like to make it work after so long.

Thanks and regards

MickTheMechanic, 2021-11-20:
Hi,

My guess is your car uses some other kind of signal for the Tacho. Do you have access to an oscilloscope to check what kind of signal you have?

JeFeRoJo, 2021-11-20:
I don't have an oscilloscope because I've never used one but I can try to get a "cheap" one like this: shorturl.at/szQ16 
It's enough?

However, in another thread here, a user has made it work on his Honda. It also added a screen.

For this reason, I was wondering if I have to edit values ​​of the code u is a circuit problem.