MickTheMechanic/Shiftlight #7: For building up the Shiftlight

sono0312, 2022-03-29:
Hello,
I am trying to build up your shiftlight for my car BMW E90 2005 which has an OBD2 port. I believe that it can give a signal RPM from PIN-9 on the OBD2 connector (according to my research).

I am still a beginner to do the sketch on the Arduino and need more time to learn/modify your sketch you posted already to match with what I want. I have checked your schematics and info of corrections on the wiring of PNP. I hope that 12-14V together with RPM can be wired from OBD2 as well.

Once I build up the Shiftlight with the test sketch you posted, is it possible to see if it is working without car? Once confirmed I want to start testing the function with the car.

Thankyou for your help.

br//sono

sono0312, 2022-04-03:
I am trying to check the parameter around PNP transistor in the circuit below. I wonder if a register 10k ohms between Collector and G are a proper value. It might be too large. If Ic should be 20mA or so the register could be 250 ohms instead of 10k ohms.

https://github.com/MickTheMechanic/Shiftlight/issues/4#issuecomment-850627599

sono0312, 2022-04-03:
> I am trying to check the parameter around PNP transistor in the circuit below. I wonder if a register 10k ohms between Collector and G is a proper value. It might be too large. If Ic should be 20mA or so the register could be 250 ohms instead of 10k ohms.
> 
> [#4 (comment)](https://github.com/MickTheMechanic/Shiftlight/issues/4#issuecomment-850627599)

I realized that 10k ohms on Corrector could be a pull-down register so it should not be a problem. Am I right?

MickTheMechanic, 2022-04-05:
It is a voltage divider to make an appropriate signal voltage

sono0312, 2022-04-05:
> It is a voltage divider to make an appropriate signal voltage

Hello,
Thankyou for the reply.
Now I was able to make a shift-light according to your information. Everything is working fine so far. To check the all the signal function I changed the frequency range to lower so that the test can be done at the lower RPM.
To check the Tach signal from OBD2, a serial monitor enabled so that I can seen the real time frequency value in response to the axel work.

Again, thankyou for your job done to make it happen!

regards

sono0312, 2022-04-16:
Found that a Line 121 starting with  "if (igfreq < onFrequency) {" and onwards need to be corrected.

===
      hasStartupSequenceRun = true;
      pixels.fill(pixels.Color(0, 0, 0));
      pixels.show();
    }
      if (igfreq < onFrequency) {
    //resets hasStartupSequenceRun to false if engine stops but Arduino remains powered, 
    //startupsequence will rerun upon restarting the engine
	hasStartupSequenceRun = false;
    }
  }
===

it should be out side of startup sequence like below.

===
    }
  }
  if (igfreq < onFrequency) {
  //resets hasStartupSequenceRun to false if engine stops but Arduino remains powered, 
  //startupsequence will rerun upon restarting the engine
    hasStartupSequenceRun = false;
===

otherwise "hasStartupSequenceRun" will never become false even the engine stops but Arduino remains powered.

sono0312, 2022-04-16:
Found that a Line 121 starting with "if (igfreq < onFrequency) {" and onwards need to be corrected.

  **hasStartupSequenceRun = true;
  pixels.fill(pixels.Color(0, 0, 0));
  pixels.show();
}
  if (igfreq < onFrequency) {
//resets hasStartupSequenceRun to false if engine stops but Arduino remains powered, 
//startupsequence will rerun upon restarting the engine
hasStartupSequenceRun = false;
}
}**

it should be out side of startup sequence like below.

**}
}
if (igfreq < onFrequency) {
//resets hasStartupSequenceRun to false if engine stops but Arduino remains powered,
//startupsequence will rerun upon restarting the engine
hasStartupSequenceRun = false;**

otherwise "hasStartupSequenceRun" will never become false even the engine stops but Arduino remains powered.

MickTheMechanic, 2022-04-16:
thanks, ill update it when I have a chance to test it out