MickTheMechanic/Shiftlight #8: tachometer wire

Baileyyyyy, 2022-05-29:
How are you sure it’s pin 9 on the obd2 connector?? every pin out i try and look at all says thier manufacturers, but i can’t find any information on what they are??

MickTheMechanic, 2022-05-29:
Pin 9 on the obd is reserved for manufacturer specific use, which on Daihatsu happens to be the RPM signal.

Baileyyyyy, 2022-05-29:
Are you able to help me find where the tacho is on a ford focus 2014 mk3? I cant find any information and the ford forums arnt any help; I know toyota also used 9 as rpm too

MickTheMechanic, 2022-05-30:
You need to do some research, what kind of ignition system you have and where you can tap into an rpm signal

baileyfpv, 2022-06-02:
Hey, found some diagrams, what do you think about the violet wire?
[image]

baileyfpv, 2022-06-02:
http://excesstext.com/tag/duratec/

there us this website too which has a full diagram of everything in the duratech engine but I cant seem to find the tacho? but that violet wire in the diagram above seems like it could work no? using the coils as the tacho signal for the less?

MickTheMechanic, 2022-06-02:
No, the violet wire isn’t the Tacho signal. It’s possible the Tacho signal is transmitted on the can-bus. 

Do your coils have two wires or three?

baileyfpv, 2022-06-02:
Unsure, Id say 2 since its a 2.0l ti-dcv

baileyfpv, 2022-06-02:
Wait Its two since thats the diagram from my engine?

MickTheMechanic, 2022-06-02:
Yeah, can’t help you if you’re unsure. I suggest you take a look, and find a full wiring diagram for the engine electrics that matches your car and post it here

baileyfpv, 2022-06-02:
Surly you can take a quick look, I assure you its pretty hard, I don't see why its so secretive about wiring

MickTheMechanic, 2022-06-02:
What’s your engine code?

MickTheMechanic, 2022-06-02:
And how much power does the engine make? I need those things so I can find the correct wiring diagram

baileyfpv, 2022-06-02:
Scratch that its TXDB 120kw

MickTheMechanic, 2022-06-02:
Is it possible it’s the XTDB? 1.6l?? I can’t find the TXDB

baileyfpv, 2022-06-02:
2.0L not exactly sure if it is but google says it’s should be the TXDB from
2014

MickTheMechanic, 2022-06-02:
What do the registration papers say? It’s important we have the right wiring diagram..

baileyfpv, 2022-06-02:
I’ll take a look tomorrow and let you know

Baileyyyyy, 2022-10-11:
There is no information that I can find unless I’m looking wrong..

closed 2023-06-03 by GH-WRX