MickTheMechanic/Shiftlight #10: NodeMCU / ESP8266 weird flashing

craigrobbo, 2022-07-10:
Hi there.

I am still pretty new to coding and trying to learn, i tried the code on my NODEMCU, the initial start up animation seems to work fine (I have not  yet connected to my engine as I am awaiting my transistors) but every 5 seconds or so all the LEDs flash red for a very brief moment, on my Arduino UNO it doesn't do this.

I was hoping to migrate to the ESP platform as I have some custom PCB boards for an ESP8266 project which would suit this and keep the physical size package really small.

Thank you 

Craig

Marius1418, 2022-09-02:
hello, 

I have the same issue with Pico board, did you manage to solve it?

MickTheMechanic, 2023-01-10:
Is anything connected to the tacho signal input when this occurs? If not then its probably just noise being registered as frequency. Try gounding the tacho pine when bench testing.

closed 2023-12-02 by MickTheMechanic