MickTheMechanic/Shiftlight #11: Please help me change the code.

A-Alikin, 2023-01-10:
Hello
To begin with, I want to express my gratitude for your work, and that you find time to help other people with their questions.
I used your design for my lexus is200. I also used 16 LEDs. However, there are some minor changes.
I changed the order of the LEDs (as in the diagram). everything works great and I love it. But each time the ignition is turned on, the initial startup animation will go on and the LEDs will light up in the order 1-2-3..16.
But in my case, this is slightly illogical and I wanted the starting animation to be in the following order: 1, 16 - 2, 15 - 3, 14 - 4, 13 - 5, 12 - 6, 11 - 7, 10 - 8, 9.

Could you help me to make changes to the code to get this order of the start animation?
Sincerely, Alexander.

[image]
[image]
[image]
[image]

MickTheMechanic, 2023-01-10:
Hey M8, nice work!

This was one of my first Arduino projects, Im glad you appreciate it and have added your own touch!

Here is how we change the code:

This part of the code contains the startup sequence:

```

 if (hasStartupSequenceRun == false) {
    if (igfreq > onFrequency) {
    //run start sequence
    //LEDs will light up, flash and light out upon starting the engine.
      for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        delay(50);
      }
      for(int a=0; a<10; a++) {
        pixels.fill(pixels.Color(0, 0, 120));
    	pixels.show();
    	delay(20);
    	pixels.fill(pixels.Color(0, 0, 0));
    	pixels.show();
    	delay(20);
       } 
       for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        }
       for ( int i = NUMPIXELS-1; i >= 0; --i) {
        pixels.setPixelColor(i, pixels.Color(0, 0, 0));
        pixels.show();
        delay(50);
      }
      hasStartupSequenceRun = true;
      pixels.fill(pixels.Color(0, 0, 0));
      pixels.show();
    }
      if (igfreq < onFrequency) {
    //resets hasStartupSequenceRun to false if engine stops but Arduino remains powered, 
    //startupsequence will rerun upon restarting the engine
	hasStartupSequenceRun = false;
    }
  }

```

These are the two interesting parts:

```

 for ( int i = 0; i < NUMPIXELS; ++i) {
        pixels.setPixelColor(i, tachColor[i]);
        pixels.show();
        delay(50);
      }
//................................................................................//
       for ( int i = NUMPIXELS-1; i >= 0; --i) {
        pixels.setPixelColor(i, pixels.Color(0, 0, 0));
        pixels.show();
        delay(50);
      }

```

These loops will squentially turn on or off one pixel followed by a delay. 

i represents the loop iteration, and NUMPIXELS represents the iteration at which the loop should stop.

To achieve the desired effect we should be able to halve the loop iterations, and count up and down simultainously to meet in the middle.

```

for ( int i = 0; i < NUMPIXELS/2; ++i) {
        pixels.setPixelColor(i, tachColor[i]); //turn on pixel counting up from 0
        pixels.show();
        int j = NUMPIXELS-1; //pixel address begins at 0, so we subtract 1 from NUMPIXELS to have the address of the last one
        j = NUMPIXELS - i; //count down from the last address each loop iteration  
        pixels.setPixelColor(j, tachColor[j]); //turn on pixel counting down from NUMPIXELS
        pixels.show();
        delay(50);
      }
//................................................................................//
      for ( int i = 0; i < NUMPIXELS/2; ++i) {
        int j = (NUMPIXELS/2)-1; //set j to one less than 1/2 NUMPIXELS
        j = j - i; //Count down from half way
        pixels.setPixelColor(j, pixels.Color(0, 0, 0));
        pixels.show();
         j = NUMPIXELS/2; //set j to 1/2 NUMPIXELS
        j = j - i; //Count upfrom half way
        pixels.setPixelColor(j, pixels.Color(0, 0, 0));
        pixels.show();
        delay(50);
      }

```

Test it out and let me know :)

A-Alikin, 2023-01-11:
Thank you fast reply.
I'll test it today and give you feedback. good day.

A-Alikin, 2023-01-11:
It works !!  I am very happy!  thank you!!  Best regards from Russia, Alexander.

[image]

MickTheMechanic, 2023-01-11:
Nice

closed 2023-01-11 by MickTheMechanic