MickTheMechanic/Shiftlight #12: Help for running on single cylinder engines

TechFreak9356, 2023-01-11:
Hi there, I came across your project of a DIY shift light and I figured it would be cool to be able to install one like this to my single cylinder motorcycle. I would like to learn more and better understand how the code works, especially in detecting the engine RPM through frequency from tach in or crank position sensor or magneto pickup, etc. 

I am trying to adapt your project to my single cylinder but I can't quite figure it out, or to be exact....not sure of how the frequency is calculated and/or measured, since the project is originally for 4 cylinder cars. I'm not that familiar with engines, and googling only made me more confused as well so I was not quite sure with my understanding of the frequency measurement.

Any help would be appreciated, thank you

MickTheMechanic, 2023-01-11:
Hi.

You need to ask yourself what will you use as a signal. How many times do you have this signal per cank rotation.

Number of signals per crank rotation * RPM / 60

So if you have a single cylinder 4 stroke then you will use the ignition coil driver as a signal, then you have one signal for every two rotations, so mathematically speaking 0.5 signals per crank rotation.

(A 2 stroke signle cylinder would have 1 signal per rotation)

Multiply by the RPM you want the shift light to be triggered at:

0.5 signals per rotation * 5000 RPM

= 2500 signals per crank rotation

Then divide by 60 to convert from signals per minute to signals per second, AKA Hertz, AKA frequency

2500/60

= 41.6 Hz

Thats how it works :)

TechFreak9356, 2023-01-11:
Thank you very much for helping out. I have edited my original post to include some more details regarding my motorcycle and my setup.

Somehow my digital dash which, as the seller said, interprets 16.6Hz @ 1000RPM for the RPM mode compatible with my motorcycle. This calibration is accurate with the stock CDI unit (which I accidentally got killed). The afternarket CDIs on the other hand, is not detected at all by the digital dash. Like I said, only the analog tachometer can then detect it, which further complicates the computation for RPM (but I understood how it's now computed - 8.33Hz @ 1000RPM).

Was it possible that my aftermarket CDI was producing way higher voltage/power than the stock and is causing the digital tach not to recognize it? or is it releasing a signal that's out of range? 

The analog tachometer cant detect the RPM signal from the RPM pin of the aftermarket CDI unit, but the digital dash can, only thing is, it's inaccurate once it gets past 4000RPM (My engine pushes 13000RPM as I verified on the analog tach connected to ignition coil positive). The opposite happens when I connect the digital dash to the ignition coil out (it has no reading at all) and the analog tach on the RPM pin of CDI (no reading).

I haven't tried to actually build the arduino yet and test it for myself. But, in case I might need to plug directly on the ignition coil positive, and the aftermarket CDI is actually pushing higher voltage through the line, will the resistors be enough to handle that higher voltage? I still have no idea what frequency the CDI RPM pin pushing out since the dash's 16.6Hz is twice that of our computed 8.33Hz @ 1000 RPM

MickTheMechanic, 2023-12-02:
You will probably need to include a diode to stop induction voltages damaging the Arduino if you connect directly to the coil.