MickTheMechanic/Shiftlight #17: Startup Sequence

Zex0208, 2023-09-18:
Hi Mick, first of all i wanna thanks you for the beautiful shift light that you shared with us. I had successfully recreated what you taught us to. I  am driving a car that is almost same engine as yours, the 3sz-ve but i am having some trouble with the start up sequence that only run 1 time when i start the car.

MickTheMechanic, 2023-12-02:
What is the problem?

Zex0208, 2023-12-02:
> What is the problem?

The startup sequence only run once. After that it won’t do the startup sequence after I started the car .

So when I disconnect and reconnect the OBD port it run again but also only once. The power seems still powered up even I turned off my car. 

How can I fix this ? By some coding ?

Thanks in advance. 🫡