MickTheMechanic/Shiftlight #18: 2009 Mustang GT help

LunixXd, 2023-11-29:
Hey Mike I stumbled across your post when looking to make things in my avionics class. I’m currently a student and have access to many things such as an oscilloscope so I thought I would attempt to make this shift light work in my car.

I’ve done some research trying to figure out how to connect this and I have remade your schematic v1.2 on a pcb board with the dip trace software  and im going to print it tomorrow at school. I was wondering how to safely connect this without blowing anything up in my vehicle, I care about it. With that being said Ive seen many people pull the tacho in from the coil driver wires as seen in this post from an official tach light company, raptor performance. ( https://www.raptorperformance.com/pages/raptor-shift-light-install-mustang-2005-2010-4-6l-5-4l-v8 ) is that where I should pull a tach signal for your circuit or should I go through via obd2, I’m not sure what Pinout provides rpm with my obd either as all companies seem to be different. I can provide photos of the circuit in a few days after I build it and give you any other vital info you need to help out. Any input is appreciated and thank you for sharing this information with us.

Ps: this is the first time I’ve ever really done anything this advanced with electrics and Arduinos so I’m learning as I go.

MickTheMechanic, 2023-11-30:
Hi,

so for a signal you just need something that is rpm relevant, you could use the signal for the ignition coil, the cam sensor, the Tacho signal from behind the dash, and so on. You just need to have a think about home many times per rotation the signal you are using triggers, and calculate the shift frequencies in the code accordingly.

The signal input on the board goes through a voltage divider to lower the voltage down to an acceptable level for the Arduino. If you are using a 5v signal then you don’t need these.

The Tacho signal on the OBD connector seems to be only Daihatsu and a few other Japanese manufacturers, consult a workshop manual/wiring diagram to see if this is also present on your car. I can’t say if American manufacturers also did this.

Please be aware that there was a mistake in the wiring diagram that is shown in README.me. There is a link there to an issue where a sketch is shown showing the correct connection of the transistor. Please take a look at that if you haven’t already.

M52byfix, 2023-12-02:
Hi friend. I'm setting up the circuit right now. But your transistor connection seems to be wrong. And my car has 6 coils. Do I need a transistor? And is there a problem with the circuit diagram? I've tried many times. There is a problem. The transistor connection is questionable. I would be very glad if you could help. I am waiting for your reply.

MickTheMechanic, 2023-12-02:
As I said earlier, the correct transistor connection is shown in issue #4

MickTheMechanic, 2023-12-02:
[image]

M52byfix, 2023-12-02:
My rpm signal is 5v. Should I still use transistors? And without a transistor, is there a difference in the circuit diagram?

MickTheMechanic, 2023-12-02:
You should be able to connect the 5v signal directly to the gpio pin without any need for a voltage divider or transistor.

MickTheMechanic, 2023-12-02:
Keep In mind that the Arduino needs to have a ground connection to the car

M52byfix, 2023-12-02:
Gpio. I didn't understand that. Maybe d2 pin?

MickTheMechanic, 2023-12-02:
Yes, gpio is just another name for the Arduino pin

MickTheMechanic, 2023-12-02:
Just connect your 5v signal directly to the d2 pin

M52byfix, 2023-12-02:
D2. Is this true. Because I threw away 2 arduino nanos. Yes, this is a shame. I tried all kinds of remedies to make it work. And the arduino nano is broken. X 🙈

M52byfix, 2023-12-02:
D2 I'm going to put a 10k resistor between the rpm signal cable and the rpm signal cable. As a precautionary measure

MickTheMechanic, 2023-12-02:
That should be ok

M52byfix, 2023-12-02:
I am grateful for your quick replies. Thank you very much

M52byfix, 2023-12-03:
> Bu iyi olmalı

Hi friend. It worked, and it worked. But sometimes it works inconsistently. I saw a video of a Honda. It has a Honda-like problem and sometimes it doesn't work at all for a while.

LunixXd, 2023-12-06:
Finally got the board complete after waiting for shipping, I will be testing it in the lab come morning, I’ll let you know how well it functions once I get it installed. but in the meantime here are photos of the board

[image]
[image]

Brown is ground, red Is power and green is signal, just for the time being. The board printed too small during the build so I readjusted it to make it work. I can’t drill holes in a straight line for the life of me so I just cut the legs of the arduino and lined them where I needed them then soldered connections.

[image]

This last photo is the board in our schematic  software, diptrace. Showing it just in case I did anything incorrectly.

LunixXd, 2023-12-06:
[image]

It works! Now when connecting it to my cool should I use the positive side for the tach in signal? When I flip the leads of the signal generator for a negative wire it doesn’t pick up anything.

LunixXd, 2023-12-07:
I had posted a photo of the schematic above, it follows the same connection as the paper drawing that mick drew

M52byfix, 2023-12-08:
[image]
[image]
[image]
[image]
[image]
I did without a transistor. But it is more advantageous to use transistors.

LunixXd, 2023-12-09:
Mick, I had it installed and here’s what happen: I turn the car on and the startup animation runs as normal, after that the lights blink red at an idle then I reved it and they turned off permanently. I uninstalled it and was going to do some troubleshooting of my own however, my question if my car is having an output from the coil as pulses per minute. Would I need to convert the pulses to a 12v signal, if so how? Or should they be fine like that after altering the code?

MickTheMechanic, 2023-12-09:
Add a Serial.print to the code to print the frequency to the serial monitor

M52byfix, 2023-12-09:
> Frekansı seri monitöre yazdırmak için koda bir Serial.print ekleyin
[image]
What is the purpose of 10k resistance. And why does it go to the gnd line. Is it necessary to put a resistor between the transistor and the D2 pin?

MickTheMechanic, 2023-12-11:
You have a 5v signal correct? If that’s the case then you don’t need any transistor or resistor. Try connecting directly to D2

the 10k and 47k resistors create a voltage divider, which drops the voltage.

MickTheMechanic, 2023-12-11:
Also make sure the Arduino is connected to the vehicle ground

LunixXd, 2023-12-11:
Hey mick, do you think this would be possible to make? It looks too simple to not attempt to especially with its high price all it does is Converts Low-Voltage/High-Current Pulse To A Clean 12-Volt Square Wave.

MickTheMechanic, 2023-12-11:
If you have a 5v signal then connect directly to D2.

What you are asking is pointless, you want to convert a 5v signal to 12v so that you can convert it back to 12v. 

Just try to connect directly to D2 if you are sure you have a 5v signal

LunixXd, 2023-12-11:
Ah I see, that makes more sense. I need to use my brain more often lol. But what doesn’t make sense is the hyperactive hz signal I’m receiving from the coil. It doesn’t even seem correlated to the rpm’s when I reved it and I can’t get a good stable reading. The circuit works just fine on the bench and that’s why I was looking at this adapter for the engine but maybe your fix may work I will have to test it down the road as school is currently closed for break. They use that adapter on my car for mechanical and electrical shift lights. There isn’t a problem with your circuit but there is one with  my vehicle and how it gets its rpm signal, thank you for taking the time to respond to me.

MickTheMechanic, 2023-12-11:
If you are using this circuit with a 5v signal then it would explain your sporadic frequencies, since you are not getting a proper high (5v) low (0v) logic. You are getting something like 0v to 2v because the voltage is being divided. So sometimes a high is being detected as a high and sometimes a high is being detected as a low.

this is why I keep insisting you test with the 5v signal directly to D2.

MarcPerformance, 2023-12-26:
> ![IMG_20231203_205236](https://private-user-images.githubusercontent.com/93797743/288975902-dd625768-fed4-4338-ba54-c0b4fa193726.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MDItZGQ2MjU3NjgtZmVkNC00MzM4LWJhNTQtYzBiNGZhMTkzNzI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4OTc1OTUxZWMzYWVmOGU4MjhkNjg0OTVkMDkyZTI5MjkwOWIwYmNjYWFmNDZjNThhYThjZmUzZjFhYTU5MmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b_vXeqJCGUp5uQhNzbQLLDLA0g_ZJdtWHeDE0bbEbDc) ![IMG_20231203_205258](https://private-user-images.githubusercontent.com/93797743/288975909-d270c11c-1718-461b-b758-4a31b55ce411.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MDktZDI3MGMxMWMtMTcxOC00NjFiLWI3NTgtNGEzMWI1NWNlNDExLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE2ZTQ0MTMxZmIwZDI5NThhODUzZDE5OGQ5NTlhYTk4ZTllNjViNjJjNDgyYWFiZjllMzQ1Yzg3ZjJmNTM0ZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nXXaQDLkpHsOKVgh5b-f9v1czUvEcOME_Z9hKxOHwF4) ![IMG_20231203_205333](https://private-user-images.githubusercontent.com/93797743/288975914-f090a4d7-abdb-4c01-9a71-b4a692029f04.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MTQtZjA5MGE0ZDctYWJkYi00YzAxLTlhNzEtYjRhNjkyMDI5ZjA0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI4YzlmNzNlNDhiMzNmZWQ1ZmFjMDM4YWZmMjcyNmNmYzRjYjM3MzFlMGMxYTM2NGQ1NTVmNTAxZTYyOTM4NDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ON4sc47a7q7cIs1L6i3QDrv7NQSDsUJAfBO4i8WX-yE) ![IMG_20231203_205345](https://private-user-images.githubusercontent.com/93797743/288975918-484fb95e-34f5-49ec-acd4-f1779b2e05fc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MTgtNDg0ZmI5NWUtMzRmNS00OWVjLWFjZDQtZjE3NzliMmUwNWZjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA0MzUwOTY0OTI1YzZiM2U0YmJkYzQyMmNjMDg3MmNhNWRlNGYxMTYxNzY0Y2QxZjU2NTI1ZWJjMjM4ZGU3OGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Olj7-sAD1Su_QU2Q2sOMzhCVFtAzBH4Ce6_ZP0UA5fE) ![IMG_20231203_205353](https://private-user-images.githubusercontent.com/93797743/288975922-09672a3f-e8a7-4237-bbd9-89b8ae5320f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MjItMDk2NzJhM2YtZThhNy00MjM3LWJiZDktODliOGFlNTMyMGYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkMDgzZWE4NTg3OTZkYmZhNmIwOGY5OWMwMjlhOWUxNDgyOWZjZjBhMTc1OTk3MGMzZjg2YzA3MTk1N2QxNTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FZr759_t_dIYxWpvz1z_WNyvWBvxRB_N5W957E8t3HM)

Hello Mike! first of all, thank you for this project. I built it, I'm going to test it on my personal car FORD FOCUS 2009 Gasoline. I mounted the transistor in the correct position.
Before I mount it, how do I test it on the work table?
 How do I apply the signal to the input when it is on the work table? Above in a post by M52byfix, in the pictures I notice a device near the dashboard clocks.... What device is it?
 
 Thank you!
Marco

M52byfix, 2023-12-26:
> > ![IMG_20231203_205236](https://private-user-images.githubusercontent.com/93797743/288975902-dd625768-fed4-4338-ba54-c0b4fa193726.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MDItZGQ2MjU3NjgtZmVkNC00MzM4LWJhNTQtYzBiNGZhMTkzNzI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4OTc1OTUxZWMzYWVmOGU4MjhkNjg0OTVkMDkyZTI5MjkwOWIwYmNjYWFmNDZjNThhYThjZmUzZjFhYTU5MmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b_vXeqJCGUp5uQhNzbQLLDLA0g_ZJdtWHeDE0bbEbDc) ![IMG_20231203_205258](https://private-user-images.githubusercontent.com/93797743/288975909-d270c11c-1718-461b-b758-4a31b55ce411.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MDktZDI3MGMxMWMtMTcxOC00NjFiLWI3NTgtNGEzMWI1NWNlNDExLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE2ZTQ0MTMxZmIwZDI5NThhODUzZDE5OGQ5NTlhYTk4ZTllNjViNjJjNDgyYWFiZjllMzQ1Yzg3ZjJmNTM0ZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.nXXaQDLkpHsOKVgh5b-f9v1czUvEcOME_Z9hKxOHwF4) ![IMG_20231203_205333](https://private-user-images.githubusercontent.com/93797743/288975914-f090a4d7-abdb-4c01-9a71-b4a692029f04.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MTQtZjA5MGE0ZDctYWJkYi00YzAxLTlhNzEtYjRhNjkyMDI5ZjA0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI4YzlmNzNlNDhiMzNmZWQ1ZmFjMDM4YWZmMjcyNmNmYzRjYjM3MzFlMGMxYTM2NGQ1NTVmNTAxZTYyOTM4NDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ON4sc47a7q7cIs1L6i3QDrv7NQSDsUJAfBO4i8WX-yE) ![IMG_20231203_205345](https://private-user-images.githubusercontent.com/93797743/288975918-484fb95e-34f5-49ec-acd4-f1779b2e05fc.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MTgtNDg0ZmI5NWUtMzRmNS00OWVjLWFjZDQtZjE3NzliMmUwNWZjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA0MzUwOTY0OTI1YzZiM2U0YmJkYzQyMmNjMDg3MmNhNWRlNGYxMTYxNzY0Y2QxZjU2NTI1ZWJjMjM4ZGU3OGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Olj7-sAD1Su_QU2Q2sOMzhCVFtAzBH4Ce6_ZP0UA5fE) ![IMG_20231203_205353](https://private-user-images.githubusercontent.com/93797743/288975922-09672a3f-e8a7-4237-bbd9-89b8ae5320f1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM2MjA0MjEsIm5iZiI6MTcwMzYyMDEyMSwicGF0aCI6Ii85Mzc5Nzc0My8yODg5NzU5MjItMDk2NzJhM2YtZThhNy00MjM3LWJiZDktODliOGFlNTMyMGYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI2VDE5NDg0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkMDgzZWE4NTg3OTZkYmZhNmIwOGY5OWMwMjlhOWUxNDgyOWZjZjBhMTc1OTk3MGMzZjg2YzA3MTk1N2QxNTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FZr759_t_dIYxWpvz1z_WNyvWBvxRB_N5W957E8t3HM)
> 
> Merhaba Mike! öncelikle bu proje için teşekkür ederim. Ben yaptım, kişisel arabam FORD FOCUS 2009 Benzinli üzerinde test edeceğim. Transistörü doğru konuma monte ettim. Monte etmeden önce çalışma masasında nasıl test edebilirim? Çalışma masasındayken sinyali girişe nasıl uygulayabilirim? Yukarıdaki M52byfix'in bir gönderisinde, resimlerde gösterge paneli saatlerinin yanında bir cihaz dikkatimi çekiyor... Hangi cihaz bu?
> 
> Teşekkür ederim! Marco

You need a frequency generator. Square wave frequency generator

MarcPerformance, 2023-12-26:
M52byfix , thanks for the quick response. I built this device from the youtube clip. Can I do these tests with it?

https://www.youtube.com/watch?v=GBmlMa-vJ8c&list=PLsp5MpqlTBRZajQEVdIYyT1N4g78S8RrR&index=36

Marco

M52byfix, 2023-12-26:
> M52byfix, hızlı yanıtınız için teşekkürler. Bu cihazı youtube klibinden oluşturdum. Bu testleri onunla yapabilir miyim?
> 
> https://www.youtube.com/watch?v=GBmlMa-vJ8c&list=PLsp5MpqlTBRZajQEVdIYyT1N4g78S8RrR&index=36
> 
> Marco
https://youtu.be/Dq0noaVAR50?si=JwWL_Y8V5ibtb3HS

This is my video. Watch

MarcPerformance, 2023-12-26:
[image]

I managed to test it on the work table!  🤩

LunixXd, 2024-01-18:
Mick, late response but the voltage was too high for the chip and I blew it to be honest, got a new one installed, works on the test bench just fine, tried it the in the car again, same results. I’m not sure where else I could pull a hz signal related to the tach via obd or the back of the gauge cluster. Is there any other way of doing this? I have no clue how my car gets its RPM signal at this point maybe the ecu under the hood??? But still at square one again what would you suggest I do?
Thx for any input

LunixXd, 2024-01-26:
It seems the coil I chose which is closest the driver is a bit fuzzy when at a flatline on the o-scope. (I don't think coil is faulty but could be a possibility.) Would a Schmitt trigger work well in combination with the board to get a clean precise output on the lights? Also, would overhauling or modifying any part of code allow compatibility between a coil connection? Simply put, my car doesn't output the right signal for the board so I have to do something before the board to clean/compute the signal.

LunixXd, 2024-01-27:
[image]
you said im supposed to be getting square waves too correct? These are the type of waves I get from the coil connecting with the o-scope.

LunixXd, 2024-03-26:
[image]
Was able to get it working and installed quite nicely, I had to go through the ecu to get a signal however ford blocks the pin (25) that you have to get to and I had to find a donor Mustang to get the right square connecting pin. To anyone else attempting this, save yourself time and go through the ecu.

closed 2024-03-26 by LunixXd