MickTheMechanic/Shiftlight #19: Compatibility with OBD2 Standard (2005 Suzuki Swift)

ElectronicallyE, 2024-01-23:
I'm trying to make your shift light project compatible with a 2005 Suzuki Swift EZ RS415 which support K-line communication known as [ISO 9141-2](https://www.suzuki-forums.com/threads/suzuki-swift-ecu-reading.277364/post-1622460) (ISO 14230-4 is backwards compatible).

Direct quote from the [Suzuki workshop service manual](https://www.allcarmanuals.com/factory-service-manual-233-Suzuki-Swift.html):
_**Data Link Connector (DLC)**
DLC (1) is in compliance with SAE J1962 in the shape of connector and pin assignment.
OBD serial data line (3) (K line of ISO 9141) is used for SUZUKI scan tool or OBD generic scan tool to communicate with ECM, Air bag SDM, immobilize control module (in ECM), BCM (Body electrical Control Module), TCM (Transmission Control Module) and ABS control module._

A few questions:
1. Could you please let me know the OBD2 standard used by your Daihatsu M100 (1998-2004) which the project is designed for?
2. Are you aware of if my car is compatible with your circuit diagram and code?

I have an oscilloscope to verify the signal type. If you would like me to verify, please let me know what I need to check.

yu835730500, 2024-03-09:
Arduino board (can use Nano, Teensy, Pro Micro, Uno....)
OBD II connector 
CAN Bus MCP2515 module

ElectronicallyE, 2024-03-11:
Thanks mate, much appreciated.

closed 2024-03-11 by ElectronicallyE