MickTheMechanic/Shiftlight #22: Startup animation restart after engine stops

LunixXd, 2024-04-03:
I’d like to get the code to preform a restart when the engine is inactive, I’d imagine I’d measure the frequency at one of the current pins and have the Arduino run a reset code. I’m not well versed in coding so I was hoping you could help lead me in the right direction to get this to work. Alternatively I could find a wire in my car that’s 12 volts switched so that it manually restarts.

LunixXd, 2024-04-06:
Also, is there a way to switch between different code presets with a click of a button?

MickTheMechanic, 2025-08-24:
That is definitely doable, I’m pretty sure the code is already there. There is a threshold frequency value, if the input frequency drops below this value it raises a flag, which triggers the start sequence one the frequency comes above the threshold again.

MickTheMechanic, 2025-08-24:
Move lines 121 - 125 to line 127 (outside the if statement) and it should work