MickTheMechanic/Shiftlight #25: Verification of circuit - Rev Light

Matheu-4AGE, 2025-08-25:
Hi there

Just a quick question for the wizard behind this Rev Light @MickTheMechanic, firstly this is a great idea and circuit!! I have been thinking of recreating it for my car for a while now and eventually I have started it.
I used the V1.2 schematics (temporarily using the 5v pin) to build the circuit and I created an Arduino code just to run the Neo pixel LED to see if the regulator will be able to keep up with the current draw. So obviously the led part of the circuit works 100%. I added a 1k pot to bring the brightness down a little to try and keep the regulator a little cooler.
But I just need to know if the transistor and cap is wired correctly because I will be trying to install this in my car after I scratch around and find a proper tach signal to use.

Any feedback would be highly appreciated!! Thanks

I have attached pics of my rough circuit, everything will look better once everything is tested and works properly.

[image]
[image]

MickTheMechanic, 2025-08-25:
Brightness of the neopixel LEDs is normally adjusted in the code.

The transistor schematic in readme.me is wrong, there is a hand drawn schematic in issue #4 with the correct one (look under closed issues), but from what I can see your one seems ok. Make sure you have the 47k and 10k resistors included in the transistor schematic, unless you are using a 5v Tacho Signal.

The capacitor is just to help smoothen voltage, not much can go wrong there. It’s probably not needed at all. I designed this a long time ago, when i didn’t really know what I was doing, that’s why all the component values are huge.

MickTheMechanic, 2025-09-01:
So the input voltage should be 0-3.3v on the Arduino, so that seems ok?

If you want a pot to adjust brightness, I’d recommend connecting the pot to the Arduino and Controlling the brightness that way. Dimming the LEDs by decreasing the input voltage might lead to unexpected behavior, since these are digital components.